voodootikigod/voodoospark #36: Extra binary required for DigitalRead and AnalogRead functions

audreylim, 2015-01-05:
I wrote a Go library for voodoospark:  https://github.com/audreylim/vspark.

While testing the library I found out that I had to add a 3rd binary (https://github.com/audreylim/vspark/blob/master/vspark.go#L182) for the DigitalRead and AnalogRead functions to work. This is only required when I'm calling the functions for the very first time. The 3rd binary should be removed after the first call or the results may not be reliable. This is not documented.

I'm not sure if I've overlooked something in my code, or if this is a requirement in voodoospark.

rwaldron, 2015-01-05:
There are two types of reads: 

- Report Once `[ 0x03 | 0x04 ]`. 
  - Responses: 
    - Digital: `[0x03, pin, value]`
    - Analog: `[0x04, pin, value]`
  - In both cases, `value` is sent as two 7-bit bytes, so expect `[action, pin, lsb, msb]`
- Report Continuously `[ 0x05 ]` (which you're not using, so this is mostly irrelevant to your question)
  - Continuous Digital Report: may send only two messages per report, each message will send `[action, port, portValue]` where `portValue` is one 8-bit value (written as two 7-bit bytes), for each "port" where `port` is either 0 or 1 and defines a group of 8 pins: 0(D0-D7), 1(A0-A7) (since analog pins can be used as digital pins). `report` is 0x03
  - Continuous Analog Report: may send up to 8 messages as `[action, pin, value]` 
    - `value` is sent as two 7-bit bytes, so expect `[action, pin, lsb, msb]`

https://github.com/voodootikigod/voodoospark/blob/master/firmware/voodoospark.cpp#L180-L185

rwaldron, 2015-01-05:
Sorry, I think I may have misunderstood your report here. Also, I apologize for writing out all of that above, it appears you already have all that under control ;)

I think there is something wrong between the `bytesExpecting`, `byteCount` and the attempt to exhaust the buffer by recalling `processInput` until there is not enough bytes left.

rwaldron, 2015-01-05:
In Spark-IO, I added two functions: `analogReadOnce` and `digitalReadOnce`, which create a 2 byte buffer (`[action, pin]`) and send to voodoospark. Here's the branch: https://github.com/rwaldron/spark-io/commit/adc1de9dc0eace17d5af4c8e56512ff92c4b6abb I cannot reproduce any unreliable behavior.

rwaldron, 2015-01-05:
It's a bug report, nothing to merge ;)

audreylim, 2015-01-07:
Ahh I figured it out - vspark was based on v2.4 of voodoospark. I noticed there were some changes to the buffer size since then. Guessed you must have tested your code against the later voodoospark version so I ran vspark against the latest version and it worked out alright. 

Thanks alot for your help @rwaldron. I've updated vspark to support the newer voodoospark and it's stable now: https://github.com/audreylim/vspark

Feel free to try it out :)

voodootikigod, 2015-01-07:
Love it!

*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.

closed 2015-01-08 by audreylim

rwaldron, 2015-01-08:
Awesome :)

Resseguie, 2015-01-08:
Created #37 and #38 to add a link to Vspark from the README and gh-pages index.html under the "reference implementations" section.