voodootikigod/voodoospark #39: Is it possible to add support for the spark.core rgb led?

canda, 2015-03-24:
I mean the rgb led located at the center of the core.
I don't know which is the best way to implement it.
Here are the docs on how it works.

http://docs.spark.io/firmware/#libraries-rgb

And this is the usage example they give
```
// EXAMPLE CODE

// take control of the LED
RGB.control(true);

// red, green, blue, 0-255.
// the following sets the RGB LED to white:
RGB.color(255, 255, 255);

// wait one second
delay(1000);

// scales brightness of all three colors, 0-255.
// the following sets the RGB LED brightness to 25%:
RGB.brightness(64);

// wait one more second
delay(1000);

// resume normal operation
RGB.control(false);
```

Resseguie, 2015-03-25:
Duplicate of #31

Resseguie, 2015-03-25:
We have some thoughts there on how we could accomplish this.

canda, 2015-03-25:
thanks @Resseguie, I will follow on the other ticket then

closed 2015-03-25 by canda