voodootikigod/voodoospark #43: Rename to particle

Snorlock, 2015-06-14:
Since they changed name some readme stuff and title needs renaming.

Resseguie, 2015-06-14:
Yes, thanks for the reminder.

We'll have to be a little careful with wording because voodoospark would work with Spark Core (still what it's called) and the Photon (at least in theory, I've not got one to test). It won't work with Electron over cellular, right? So we can't just say "particle devices". 

This is just a note to make sure we think it through.

BrianGenisio, 2015-08-27:
We just did this for [spark-io](https://github.com/rwaldron/particle-io) (now particle-io).  At a very minimum, the readme for this project should be updated with the new CLI and branding changes.  I'll send a PR.

Should the project name change?  If not voodooparticle, what should it be?

Resseguie, 2016-01-23:
It looks like this has been taken care of with @BrianGenisio and @lynnaloo updates. I'm going to close and if you see any we missed additional PRs welcome.

closed 2016-01-23 by Resseguie