voodootikigod/voodoospark #57: Voodoospark issues when using multiple Digital Inputs

alfredoborrero, 2016-03-08:
Hi Chris,

First of all I would like to thank you for developing the firmware for Spark Core/Photon "**voodoospark.cpp**" this is a beautiful and elegant code. 

I was implementing voodoospark.cpp and I was using: 3 Arduino Boards + 2 Photon Boards and attaching multiple sensors (Switches, Buttons and Motion Sensors) to each micro-controller. And I noticed that the Firmware was not calculating correctly the "portValues" in order to report them to the Node Application. 

And I am talking specifically line #484: **_portValues[k] = (portValues[k] | pinValue) << i;_**

Instead of using that, i used: **_portValues[k] = portValues[k] + pinValue * factor;_**

Where _**Factor**_ is _**Math.pow(2,k)**_

The issue was present every time I apply "High" status to more than one digital input. In my case I had 3 digital inputs. If I apply "High" to only one digital input at a time it will work perfectly, however when I press the button for lets say 3 Digital Inputs it will not calculate the right value. I think it has to do with the shift operator (<<). So I decided modify the firmware I little bit and share it with you.

Would you be able to look into that and let me know if this make sense? Have you head about any other user having similar issues? This drove me crazy for a week and I have to debug pretty much every module that I was using/including (Johnny Five, Socket IO, EthernetStandardFirmata and Voodoospark)  

Regards,

Alfredo Borrero

rwaldron, 2016-03-09:
> portValues[k] + pinValue * factor; Where Factor is Math.pow(2,k)

This is definitely not right—but looking at the current operation in voodoospark, it looks like I made a mistake in precedence. Instead of: 

```
portValues[k] = (portValues[k] | pinValue) << i;
```

it should be: 

```
portValues[k] = portValues[k] | (1 << i);
```

Or better... 

```
portValues[k] |= 1 << i;
```

Let's say D0, D1, D2, D3, and D4 are all reporting and presently all HIGH. To represent that in a single 8-bit byte, it would look like: `0b00011111`, where bits 1-5 are on. For every HIGH pin, shift a 1 into its bit place.

Here's a simulation in JS: 

```js
function digitalRead(pin) {
  // simulate HIGH on pins 0-4
  return pin < 5 ? 1 : 0;
}

var portValues_a = [ 0, 0 ]; // this will store my version
var portValues_b = [ 0, 0 ]; // this will store your version
var reporting = [1, 1, 1, 1, 1]; // reporting on D0, D1, D2, D3, & D4

for (var k = 0; k < 2; k++) {
  var shouldSend = false;

  portValues_a[k] = 0; 
  portValues_b[k] = 0;

  for (i = 0; i < 8; i++) {
    pin = (k * 10) + i;

    if (reporting[pin] == 1) {
      shouldSend = true;
      pinValue = digitalRead(pin);

      if (pinValue) {
        portValues_a[k] |= 1 << i;
        portValues_b[k] = portValues_b[k] + pinValue * Math.pow(2, k);
      }
    }
  }
}

console.log(portValues_a);  // [ 31, 0 ] 0b00011111
console.log(portValues_b);  // [ 5, 0 ]   0b00000101 <-- This would look like D0 and D2 are high, but no others. 
```

Either way, there IS absolutely a bug and I'm ashamed to learn that it's been there Nov. 2014. I'm further shocked that this went un-noticed, considering the amount of people and projects that I know have worked with voodoospark and built things with Johnny-Five and Particle-IO.

I will have a fix shortly.

closed 2016-03-09 by rwaldron

rwaldron, 2016-03-09:
You can test the latest firmware changes with this: 

```js
var Particle = require("../lib/particle");
var board = new Particle({
  token: process.env.PARTICLE_TOKEN,
  deviceId: process.env.PARTICLE_DEVICE_ID,
});

board.on("ready", function() {
  var values = [];
  Array.from({length: 8}, (_, index) => {
    values.push(0);
    this.pinMode(`D${index}`, this.MODES.INPUT);
    this.digitalRead(`D${index}`, function(value) {
      values[index] = value;
    });
  });

  setInterval(function() {
    console.log(values.join("-"));
  }.bind(this), 100);
});

```

alfredoborrero, 2016-03-09:
Thanks man, that was an elegant solution. I will continue my development ... This was driving me crazy for a week :-)