voodootikigod/voodoospark #62: I2C sensor (BME280) only working when DEBUG mode is active

beldar, 2017-09-24:
Hi,

I'm using j5 to control a Particle Photon, and I'm using a temperature/humidity sensor ( http://johnny-five.io/examples/multi-BME280/ ), but I've noticed it doesn't work with the latest firmware ( V4.0.0 ).

While trying to debug the problem I activated the `DEBUG` variable at the top of the code, and then magically it started working. I've reproduced it multiple times.

Not sure why this happens but I'm pretty sure its a bug. Could you have a look?

Thanks.

rwaldron, 2017-09-26:
Sounds like there's a timing issue between the BME280 and the Photon, the only thing that `DEBUG` would introduce is a minor time delay. I'm traveling this week, so I won't be able to look at this until at least Monday. Maybe @BrianGenisio can take a look sooner?

BrianGenisio, 2017-09-26:
I feel really bad about it, but I'm so disconnected from this project and hardware hacking in general (I haven't touched a breadboard in about 10 months) .  I don't think I'll be of any value.  So sorry :(

beldar, 2017-09-27:
@rwaldron can anyone else look at it? Or could you give me some pointers of what to look for to have a try myself?

Thanks

rwaldron, 2017-09-27:
>  Or could you give me some pointers of what to look for to have a try myself?

My first thought: is the time cost of printing debug messages providing some delay that the BME280 actually needs to behave correctly? It's not unheard of for a sensor to have some kind of waiting period between writing the register that will be read and then getting the bytes from the wire. There are `DEBUG` blocks between those two operations. 

... But I've looked at BME280 drivers from Particle, SparkFun and Adafruit and none of them call out such a delay. Incidentally, no such delay exists in StandardFirmata (which is similarly implemented).

So here's what I do first if I were to sit down to this right now: 

1. Set `DEBUG` to `1`
2. Comment out _ALL_ of the `DEBUG` blocks in https://github.com/voodootikigod/voodoospark/blob/cfececc76849fb5d2439a17bd837749dcf44edf1/firmware/voodoospark.cpp#L287-L357
3. Compile, upload and test

beldar, 2017-09-29:
Actually I'm traveling myself this weekend, so I think is better if you could take a look on Monday.

Thanks!

beldar, 2017-10-04:
@rwaldron have you had the chance to have a look at this?

Thanks

rwaldron, 2017-10-05:
I spent some time updating my Photon and working to reproduce the issue today, which i was able to achieve. I will try to work on this tomorrow as well.

rwaldron, 2017-10-10:
Sorry, I haven't had a chance to work on this since I last posted.

rwaldron, 2017-10-10:
Trying to squeeze in some time right now... With `DEBUG 1`, I've gotten as far as testing against the source with each DEBUG block in `readAndReportI2cData` commented out, one at a time and then incrementally all of them. I've now commented out all DEBUG blocks in `readAndReportI2cData` and I can observe that no logging is coming from my serial monitor, but the JS program runs just fine and outputs the degrees in C: 

```js
const {Board, Multi} = require("johnny-five");
const Particle = require("particle-io");
const board = new Board({
  io: new Particle({
    token: process.env.PARTICLE_TOKEN,
    deviceName: "squeamish-captain",
  })
});

board.on("ready", function() {
  console.log("CONNECTED");

  const now = Date.now();
  const bme = new Multi({
    controller: "BME280",
  });

  bme.on("data", () => {
    console.log(bme.thermometer.C);
  });
});
```

Not sure what to probe next, but I'll report when I know more.

beldar, 2017-10-10:
Maybe some pointers on how I'm using it?

```js
const multi = new five.Multi({
    controller: 'BME280',
    address   : 0x76,
    freq      : 250
  });

//In a different part of the code I have a setInterval 
//and I get the data from the sensors directly. Something like this:

setInterval(() => console.log(multi.thermometer.celsius), 1000);
```

rwaldron, 2017-10-10:
I've determined that several arbitrary combinations of commented out DEBUG blocks will cause reporting to stop. I cannot figure out what they have in common.

> address   : 0x76

Why 0x76? Can you link me to the exact board you have?

beldar, 2017-10-10:
I found it using a I2C scanner, these boards can be on 0x76 or 0x77 depending on a jumper, so I'm fixing it to be sure

rwaldron, 2017-10-11:
> these boards can be on 0x76 or 0x77 depending on a jumper, so I'm fixing it to be sure

Right—I only asked because mine defaults to 0x77. It's also definitely unrelated, since you said that it _does_ work with `DEBUG 1`. 

Here's another thing I've found: https://community.blynk.cc/t/bme280-sensor-and-particle-photon/7473/13 This person's issue sounds _exactly_ like ours: 

> I tried to use the Adafruit library with the BME280 sensor but had no luck. Somehow I had a timing problem which I gave up on

And then they mention using `CE_BME280.h`, so I'm going to look there next

rwaldron, 2017-10-11:
Update!

> a timing problem

This got me thinking about _why_ a couple of `Serial.print()` calls might "correct" a timing issue, and _of course_ those operations have _some_ time cost! With that in mind, it occurred to me that the BME280 likely has some timing sensitivity between each operation to set the register and then read the register, ie. read, report, delay, read, report, delay (and so on). There is a sampling interval imposed by voodoospark on all continuous reads, but it was likely too high frequency for the Photon's I2C bus. Using a feature in Johnny-Five that is not documented (because it's poorly named and I was never totally sure I wanted to expose it as "recommended"), I was able to successfully read from the BME280 with the firmware's `DEBUG` set to `0`:

```js
const {Board, Multi} = require("johnny-five");
const Particle = require("particle-io");
const board = new Board({
  io: new Particle({
    token: process.env.PARTICLE_TOKEN,
    deviceName: "squeamish-captain",
  })
});

board.on("ready", function() {
  console.log("CONNECTED");

  const now = Date.now();
  const bme = new Multi({
    controller: "BME280",
    // Secret optional microsecond I2C register set & register read delay
    delay: 1000
  });

  bme.on("data", () => {
    console.log(bme.thermometer.C);
  });
});
```

beldar, 2017-10-11:
Interesting! Is there a way to fix the timing issue on voodoospark? Or would I need to use that `delay` thing on J5? I'm a bit worried they will deprecate it and then the fix is gone :(

rwaldron, 2017-10-11:
> I'm a bit worried they will deprecate it and then the fix is gone :(

I commit to you now that I will never deprecate or remove this property :)

beldar, 2017-10-11:
Hahah! That's so funny I had no idea, well I guess that settles it then! Thank you for your help!!
I'm going to use that `delay` and see if everything works

beldar, 2017-10-17:
Seems to work fine with the `delay` so I'll close the issue!

closed 2017-10-17 by beldar