voodootikigod/voodoospark #64: servoWrite enhancements

dtex, 2018-02-02:
## The short version
servoWrite only works with single byte values. Johnny-Five will soon need the ability to send microSecond values to servoWrite which requires two bytes.

## The long version
I am working on changes to Johnny-Five that add support for 90°, 360° and multi-turn (winch) servos. The changes will also allow us to have finer control over servo positioning. The core feature required for adding this is the ability to send values in microseconds instead of degrees. The means the value passed could be as high as 2400 (or maybe even a little higher).  That requires either two bytes for the value or a variable length value.

In firmata (Arduino's servo.cpp actually) the value that is passed is evaluated. If it is < 544 then servo.cpp assumes it is degrees on a 180 degree servo. If the value is higher it assumes microseconds. The 544 value is arbitrary and weird but it is what it is.

I have been working on adding this to all the Johnny-Five IO plugins one-by-one to get them to functional parity with firmata.js and that's why I'm here now.

I'm happy to get the PR in here and particle-io, but I want to do it in a way that is non-disruptive for other users. 

So my question is: **Does anybody even use voodoospark without particle-io?** If not, I can change both to use two value bytes for servoWrite and keep it simple. If other people are using voodoospark without particle-io I'll need to make servoWrite work with either 1 or 2 values bytes.

dtex, 2018-02-02:
Thinking a bit more, it is also possible that a user could have an old version of voodoospark flashed and a new version of particle-io, so maybe the right approach is to add the ability to know the version of voodoospark via a capabilities query and then send the correct one or two byte value (or throw if the user is trying to write microSeconds and it is not yet supported by the currently installed  firmware).

dtex, 2018-03-14:
Okay, here's what I'm gonna do:

Since values greater than 180 are invalid for servo.write, if the value is > 192 (bits 1 & 2 on) then we are passing a variable length value. The length of the value is given in the last few bits. Then we just call cachebuffer again and get our value bytes.

Clients that are not aware of the variable length option will still work as-is with this new version.

If uS are sent to a device running <= 4.0.0, the servo will move to 180 degrees and there will be garbage bytes in the buffer. Clients that want to send values in uS will need to clearly document that Voodoospark 4.1.0 (presumably) is required. 

This part worries me since pretty much anything could happen if those extra bytes were parsed as a command. Any ideas on how to flush those extra bytes?

dtex, 2018-03-14:
If the first byte of the variable length value is outside the action range (> 0x46) I think the Particle will just stop processing commands altogether. At least it wouldn't do some [random unexpected thing](http://www.independent.co.uk/news/world/europe/worker-killed-by-robot-at-volkswagen-car-factory-10359557.html).

dtex, 2018-04-05:
I tested the variable length thing and it does indeed do nothing, but the processor on the Particle gets pegged and who knows how long it could run or how hot it could get so I'm not going to do that.

I'm just gonna break it. Particle-io v.next will require voodoospark v.next and vice-versa.

I'm also going to add in QUERY_FIRMWARE to help avoid future pickles like this one. That means expanding the ACTION_RANGE to 0x79 with a big list of zeros in the bytesToExpectByAction array. We need to open this up to support stepper one day anyway.

rwaldron, 2018-04-05:
> I'm just gonna break it. Particle-io v.next will require voodoospark v.next and vice-versa

I support this

dtex, 2018-04-07:
I made this way harder than it needed to be. I had assumed that all the action values matched the firmata protocol. Some do, but others do not. This means I have the option of adding a new command that falls within the ACTION_RANGE and is exclusively for writing in microseconds.

PR incoming.

PR #65 (2018-04-16): Add SERVO_WRITE_MICROSECONDS
Adds a new command to allow for passing in microseconds instead of degrees.

Closes #64
closed 2018-04-16 by rwaldron